bbogush/nand_programmer #2: USB device not found.

repairman0908, 2019-04-19:
serialprint:
NAND programmer ver: 2.1.0
JTAG init...done.
LED init...done.

I tested on win7 and showed unrecognized USB devices. What went wrong. Can you help me?

bbogush, 2019-04-19:
Hi repairman0908,
Do you have "USB init...done" and "CDC init..done" in the console log? If yes then USB initialization is done and probably you need driver for Windows. I will look for it and share a bit later.
In general the programmer should work out of the box on Ubuntu.

repairman0908, 2019-04-19:
Hello，bbogush， 
Thank you for your reply！The following information is displayed only at power on or reset:
NAND programmer ver: 2.1.0
JTAG init...done.
LED init...done.
My circuit runs on the breadboard.

bbogush, 2019-04-19:
I can reproduce the same issue in 2 cases:
1. The driver is not installed. In this case you should see "STMMicroelectronics Virtual COM port" in Windows "Device Manager" as "Unknown device". I added official driver from STM to release folder  https://drive.google.com/drive/u/0/folders/1VHw-djlLal3cvaylapG-oUkLmzuEQM77
2. USB HW issue. The power connection is ok but no D+ & D-. You should check connection of mentioned lines from USB connector to MCU, also pull-up of D+ to 3.3V over 1.5R.

repairman0908, 2019-04-23:
Hi ,bbogush,
My board is working!   : )
I can read the chip ID.
But i received a error Msg : "programmer send error :104"  (when erase the chip.)

bbogush, 2019-04-24:
Hi,
Glad to hear you resolved USB issue. 
The 104 error on erase means that programmer fails to read bad blocks from a NAND chip. Could you please provide more details about your setup? 
In particular:
- Which NAND chip are you using? Is it  K9F2G08U0C?
- Which OS the program is running on? Is it Linux or Windows?

repairman0908, 2019-04-24:
Hi，bbogush,
Thank you for your patience.
1, the chip is  K9F1G08U0E.
2, The OS is Debian9 on virtual machine.
Do you have a windows program?

bbogush, 2019-04-24:
K9F1G08U0E is compatible with K9F2G08U0C which I've tested. So only "Size" should be adjusted in "Settings"->"Chip database" to 134217728 (128MB). Anyway the error is not related to this. 
I am not sure it  works on virtual machine thereby better to check with normal OS. Currently there no version for Windows but I am working to provide it in the next release. 
Usually the error 104 indicate HW issue. I would double check connection from MCU to NAND chip with multimeter according to schematic (test for continuity). 
If you are using TSOP-48 socket it may be source of errors because it does not always provide reliable connection. A NAND chip should be clean and adjusted properly in socket.

bbogush, 2019-04-28:
Hi,
You can try initial version for Windows https://drive.google.com/drive/u/0/folders/1VHw-djlLal3cvaylapG-oUkLmzuEQM77. Windows 7/8 requires installation of driver. Also correct COM port need to be set in "Settings"->"Programmer".

bbogush, 2019-05-09:
Hi,
There are some pre-compiled binaries of CFE bootloader https://github.com/shsum/T3200M/tree/master/bcm963xx_router/targets/cfe. I am not sure which one is suitable for the router flash, maybe cfe63138rom.bin.
It looks the package contains only open source code so no guarantee it is compilable and run without proprietary part. I have tried as follows and it failed at some stage:
```
mkdir ~/dev
cd ~/dev/
git clone https://github.com/shsum/T3200M.git
sudo ln -s /home/<user>/dev/T3200M/toolchains /opt/toolchains
sudo ln -s /home/<user>/dev/T3200M/bcm963xx_router /opt/bcm963xx_router
 sudo apt-get install liblzo2-dev uuid-dev
cd ~/dev/T3200M/bcm963xx_router
make --trace PROFILE=963138BGW SHELL=/bin/bash
```

repairman0908, 2019-05-10:
Thank you for your help, I have tried all the CFE, they can not start my router.
There is also source code on the Netgear website. I don't know what is different. here[https://www.downloads.netgear.com/files/GDC/2649_GPLv1.html](url) Ctrl+f search v7610 can be found it.
Thank you very much!

bbogush, 2019-05-11:
Hi,
There is open source package for V7610.  The previous T3200M package was for different router based on the same Broadcom platform. It can be compiled similar to T3200M
"make PROFILE=V7610_TR181 SHELL=/bin/bash" but it also has compilation issues. Even if compiled I think it will not be functional because of missing proprietary code. There are also binaries for cfe you can try. You should have serial console connected to the router to see they work.
Also one of the methods is to clone flash from the same router.

repairman0908, 2019-05-12:
Hi,
Thanks for your advice. I used a serial console, but no information was printed. Because I can't find the same router, I just thought of compiling the source code. Anyway, thank you very much for your help.

bbogush, 2019-06-02:
As no other issues with the programmer I am closing it. If you have any other questions, please feel free to ask.

closed 2019-06-02 by bbogush