bbogush/nand_programmer #7: Ask some questions

leovs, 2020-05-14:
Serial info：

NAND programmer ver: 3.1.0
JTAG init...done.
LED init...done.
USB init...done.
CDC init...done.
ERROR: NAND read bad block info error at 0x0

chip：ESMT  F59L2G81A

bbogush, 2020-05-14:
This error in most cases means bad hardware connection with NAND or NAND is broken. The sockets adapters are not usually provide reliable connection. It requires few tries to properly put NAND. To be sure I use soldering adapters. After soldering the adapter with NAND needs to be carefully cleaned from flux.
You can also try Settings -> Programmer -> Uncheck "Skip bad block" option in order to check that chip cannot be read. 
Are you able to read correct chip ID - c8 da 90 95 44 ?
Have you compiled the latest source or used 3.1.0 release binaries?

leovs, 2020-05-14:
Thank you for your reply
Binaries used in version 3.1.0
chip ID ：0XFD 0XFD 0XFD 0XFD

leovs, 2020-05-14:
Nando running Windows 10
The default chip database does not have f59l2g81a
add :
F59L2G81A 0x800 0x20000 0x10000000 0x40 20 12 5 10 10 12 12 12 5 5 5 25 25 20 3 2 0x0 0x30 - 0x90 0xff 0x80 0x10 0x60 0xd0 0x70 0

leovs, 2020-05-14:
Uncheck "Skip bad block" option

info:"Bad block at 0x0XXXXXXXXXX size 0x00020000" 
more
All failed

bbogush, 2020-05-14:
The latest code contains fixes and default support of f59l2g81a. I will make new release today so you can check.

bbogush, 2020-05-14:
The new 3.2.0 release of stm firmware and windows software is available for download.  Although SW contains fixes the chip ID 0XFD 0XFD 0XFD 0XFD in 99% shows NAND connection problem with programmer.

leovs, 2020-05-15:
Thank you
still:
Error: "Programmer sent error: (104) Failed to read chip"

maybe NAND connection problem，I'll check again

leovs, 2020-05-19:
It seems that the data can be read normally, but I don't know if it works correctly

Info: Connected to programmer
Info: Firmware version:  3.2.0
Info: Detecting chip ...
Info: ID 0xE8 0xFA 0xB0 0xB5 0x64
Info: Chip not found in database
Info: Configuring programmer ...
Info: Programmer configured successfully
Info: Reading data ...
Info: "Skipped bad block at 0x00000000 size 0x00020000"

bbogush, 2020-05-19:
There still problem with connection, the ID should be 
0xC8 0xDA 0x90 0x95 0x44
but is
0xE8 0xFA 0xB0 0xB5 0x64
some pins are connected correctly but others not
0xC8 - 0b11001000
0xE8 - 0b11101000
Please check FSMC_D0 - FSMC_D7

bbogush, 2020-06-08:
Closing. Please reopen the issue if you still have any questions. 
Regards

closed 2020-06-08 by bbogush