bbogush/nand_programmer #8: Qt compilation issues

enilfodne, 2020-06-01:
Hi,

While porting the firmware, i tried to compile the host application, but i got the following error:

g++ (10.1.0)
``
error: flexible array member ‘RespHeader::data’ not at end of ‘struct<unnamed>’
``

similar issue with clang++(10.0.0)

``
error: field 'header' with variable sized type 'RespHeader' not at the end of a struct or class is a GNU extension
``

It seems older versions had a bug, where they didn't properly detect this issue.

Even thought `data` is the last member in `RespHeader`, it has to be last member as well in every other struct using `RespHeader`.

bbogush, 2020-06-03:
Hi,
Thank you for reporting the issue. Probably it is just warning but due to compilation flags "QMAKE_CXXFLAGS += -std=c++11 -Wextra -Werror" it is reported as error. Anyway it should be fixed.

enilfodne, 2020-06-03:
Hi,

Thank you for taking the time to check on this issue.

I think the compiler (at least in recent versions) is classifying it as an error. Simple test by omitting `-Wall` and `-Wextra`, produces the following output:

```
g++ -c -pipe -std=c++11 -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -isystem /usr/include/qt -isystem /usr/include/qt/QtWidgets -isystem /usr/include/qt/QtGui -isystem /usr/include/qt/QtSerialPort -isystem /usr/include/qt/QtCore -I../build/moc -I../build/ui -I/usr/lib/qt/mkspecs/linux-g++ -o ../build/o/unix/main.o main.cpp
In file included from writer.h:9,
                 from programmer.h:13,
                 from main_window.h:9,
                 from main.cpp:6:
cmd.h:130:13: error: flexible array member ‘RespHeader::data’ not at end of ‘struct<unnamed>’
  130 |     uint8_t data[];
      |             ^~~~
```

AFAICS in `gcc`, flexible array members are a GNU [extension](https://gcc.gnu.org/onlinedocs/gcc/Zero-Length.html) and by the looks of it, its not identical to the C99 flexible array definition.

bbogush, 2020-06-03:
Fixed. 
Commit e843822c74e161dca21974db55ec03c9065c5d3c

closed 2020-06-03 by bbogush

enilfodne, 2020-06-03:
Thanks for the quick fix

Does the firmware needs an update as well, because there are references to the data field in multiple instances in `nand_programmer.c` and i wasn't sure if its deprecated or still in-use.

np_write_data_cmd_t:
[np_cmd_nand_write_data](https://github.com/bbogush/nand_programmer/blob/master/firmware/programmer/nand_programmer.c#L743)
[np_cmd_nand_write_data](https://github.com/bbogush/nand_programmer/blob/master/firmware/programmer/nand_programmer.c#L776)
[np_cmd_fw_update_data](https://github.com/bbogush/nand_programmer/blob/master/firmware/programmer/nand_programmer.c#L1298)
[np_cmd_fw_update_data](https://github.com/bbogush/nand_programmer/blob/master/firmware/programmer/nand_programmer.c#L1324)

np_resp_t:
[_np_cmd_nand_read](https://github.com/bbogush/nand_programmer/blob/master/firmware/programmer/nand_programmer.c#L980)

bbogush, 2020-06-03:
I think the firmware needs to be refactored as well. Though the same version of gcc does not give any warning unlike g++. Since no issues discovered so far it has low priority.

enilfodne, 2020-06-03:
There shouldn't be any errors in the firmware, because its all C and flexible arrays are part of the C99 standard, the issue was with flexible arrays in C++ and their support through a GNU extension. But my understanding may be wrong.

I tested out the new commit and it seems that something isn't working right as the `Read ID` command no longer returns the Chip ID.

bbogush, 2020-06-03:
I've tested on my board Read ID works ok. Are you sure it is regression? Was it working before? If you revert last commit does it fixes the issue?

enilfodne, 2020-06-04:
Applied the patch and the ID that's usually returned (5 bytes) is gone from the log window and underneath the select box to choose a chip from chipdb. Trying to read the chip, but the program returns a timeout error, where in fact the data keeps coming.

Using the pre-compiled versions seems to bypass all of that.

bbogush, 2020-06-05:
Unfortunately cannot reproduce this issue. Maybe problem is related to different compiler/Qt version.

enilfodne, 2020-06-06:
Just tried and successfully compiled the application with gcc 5.5.0 without the latest commit. I think reverting it might be best to avoid potential issues as it didn't solve the initial problem for gcc 10

bbogush, 2020-06-08:
ok, reverted it.

danyhm, 2021-01-14:
I have this problem too.

i've developed the ECC and reset functions in the firmware and qt however because of the same error I can't built the correct qt app. also installing gcc-5 is not possible in the latest linux versions. I tried to change it but then saw a lot of other sections have to be modified as well.

bbogush, 2021-01-14:
> have this problem too.
>
> i've developed the ECC and reset functions in the firmware and qt however because of the same error I can't built the correct qt app. also installing gcc-5 is not possible in [...]

Try to compile without -Wextra -Werror in qt.pro "QMAKE_CXXFLAGS += -std=c++11 -Wextra -Werror"

danyhm, 2021-01-14:
> Try to compile without -Wextra -Werror in qt.pro "QMAKE_CXXFLAGS += -std=c++11 -Wextra -Werror"

didn't work. never mind i fixed the code in another way. now it compiles with gcc10 and latest Qt. also the firmware is fixed to reflect the changes. I'll make a PR after i do more checks