bbogush/nand_programmer #14: Reset button

alexgrach, 2020-09-30:
It would be nice to have reset button. I have to use a wire to reset before flashing over SWD by st-link.

bbogush, 2020-10-03:
HW reset button is rarely used and can be replaced by cable unplug-plug action. Thereby I would not add additional component to PCB design which may increase its size.

closed 2020-10-03 by bbogush

alexgrach, 2020-10-07:
I've added NRTS just like BOOT0 and made some changes in schematics to make it look better and readable.
[image]
[nand_programmator.zip](https://github.com/bbogush/nand_programmer/files/5343890/nand_programmator.zip)