bbogush/nand_programmer #17: Error detecting TC58NVG1S3E

alexgrach, 2020-10-01:
I've tried to read 2 different NAND ICs. No success.

Info: Connected to programmer
Info: Firmware version:  3.3.0
Info: Detecting chip ...
Info: ID 0x8F 0x8F 0x8F 0x8F 0x8F
Info: Chip not found in database
Info: Detecting chip ...
Info: ID 0x8F 0x8F 0x8F 0x8F 0x8F
Info: Chip not found in database

bbogush, 2020-10-03:
In 99% it is hardware problem:
- no connection between STM and NAND, for example, some pins of STM are not soldered, NAND pins are not soldered or bad connection in TSOP socket.
- stray capacitance if flux is not cleaned well after soldering.

alexgrach, 2020-10-06:
I've resoldered my chip and it worked, thank you! 
chip_db to support this IC: 
`
TC58NVG1S3E, 2048, 131072, 268435456, 64, 20, 12, 12, 10, 10, 12, 12, 12, 5, 5, 5, 25, 25, 20, 3, 2, 0, 48, -, 144, 255, 128, 16, 96, 208, 112, 0, 152, 218, 144, 21, 118
`

bbogush, 2020-10-10:
Glad to hear it works for you. 
Thank you for new chip. I've added it database.

closed 2020-10-10 by bbogush