bbogush/nand_programmer #18: Reset pad

alexgrach, 2020-10-11:
I've added NRTS just like BOOT0 and made some changes in schematics to make it look better and readable.
[image]
[nand_programmator.zip](https://github.com/bbogush/nand_programmer/files/5343890/nand_programmator.zip)

_Originally posted by @alexgrach in https://github.com/bbogush/nand_programmer/issues/14#issuecomment-705198990_

bbogush, 2020-10-11:
I appreciate your contribution. The change looks really nice but currently I have some doubts whether to include it. The reason is that most programmers don't provide HW reset and I don't see use cases for it because this can be done by power off-on.

alexgrach, 2020-10-11:
I see. It seems to me that reset is needed when you update programmer FW from previous version (from 3.1 to 3.3 in my case). With new IC st-link does not ask to boot under reset. If you do like schematics I can delete reset pad (or you can do it) pcb won't suffer.

alexgrach, 2020-10-11:
[nand_programmator.sch.zip](https://github.com/bbogush/nand_programmer/files/5361173/nand_programmator.sch.zip)

bbogush, 2020-10-11:
I think now I understand. Sometime STM cannot be programmed because it has disabled SWD pins for some reason and the only way is too "connect under reset". The RST pads are required. Would you like to create git pull request with your original changes so you will be mentioned as project contributor by github? If not I can apply changes.

alexgrach, 2020-10-11:
I've never done this before but I'll try. #19

bbogush, 2020-10-12:
The changes are merged. Thank you for contribution.

closed 2020-10-12 by bbogush