bbogush/nand_programmer #23: Can you add support for STM32F103ZET6

mlpzaq9998, 2020-12-06:
[1、STM32F103ZE核心板--原理图-1807M.pdf](https://github.com/bbogush/nand_programmer/files/5648222/1.STM32F103ZE.--.-1807M.pdf)

mlpzaq9998, 2020-12-06:
[image]

Run NANDO to display the （112）wrong data length

mlpzaq9998, 2020-12-06:
[image]

bbogush, 2020-12-06:
Hello,

I see STM32F103ZET6 should be compatible with NANDO's STM32F103VCT6. 
Regarding error, the version of firmware was read correctly so USB communication works. The reason of error is that NANDO  firmware received not enough or to many data in configuration command. The UART log can show more information. Are you using released firmware and PC program provided in GoogleDriver?

mlpzaq9998, 2020-12-06:
using released PC program provided in GoogleDriver，The firmware is GitHub
How to eliminate errors?

bbogush, 2020-12-06:
The firmware and PC program should be the same version because communication protocol sometimes changes. So you should use the released PC program and firmware from google driver or checkout git tag of the same version, i.e, git checkout 3.3.0.

mlpzaq9998, 2020-12-06:
Use the same version of the checkout git tag，NANDO display

Info: Connected to programmer
Info: Firmware version: 3.3.0
Info: Detecting chip ...
Info: ID 0x8F 0x8F 0x8F 0x8F 0x8F
Info: Chip not found in database
Info: Detecting chip ...
Info: ID 0x8F 0x8F 0x8F 0x8F 0x8F
Info: Chip not found in database

It should be a hardware problem

mlpzaq9998, 2020-12-06:
Can the new firmware include HEX

bbogush, 2020-12-06:
0x8F 0x8F 0x8F 0x8F 0x8F pattern shows hardware problem - unconnected pins or high stray capacitance.
  
Yes, hex will be included, also it can be received by following conversion, but haven't tried if it works.
../../compiler/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-objcopy --input-target=binary --output-target=ihex obj/fw.bin obj/fw.hex

bbogush, 2020-12-22:
Closing old issues. Please reopen if you still have questions.

closed 2020-12-22 by bbogush