bbogush/nand_programmer #24: How to flash firmware using usb serial cable?

bra1nslayer, 2020-12-15:
Hi,
I have FT232RL same board as here: https://www.sparkfun.com/products/12731
[image]
[image]

Is it possible to flash fimware nando_fw_v3_3_0.bin via arduino ide?Or another program suggestions?
Thanks in advance, very useful programmer btw.

bbogush, 2020-12-15:
Hi,
The best option is to use stlink programmer. You can try to use build-in USB/USART bootloader. For this you need to install stm32 tool https://www.st.com/en/development-tools/stm32cubeprog.html# and after that power on board with connected BOOT0 pins to load internal DFU bootloader. Though, I've just tried and this method for some reason does not work for me, device is not recognized by Windows.

bra1nslayer, 2020-12-16:
Hi,
Uploaded firmware using serial cable but neither windows and linux recognizes as usb device.For firmware upload i have used stm32cube for windows and stm32flash for linux.Writing firmware was successful but as i said it cannot detect as usb device.Here is some info:
[image]

```

[  409.635435] usb 1-4: new full-speed USB device number 2 using xhci_hcd
[  409.763420] usb 1-4: device descriptor read/64, error -71
[  409.999461] usb 1-4: device descriptor read/64, error -71
[  410.235499] usb 1-4: new full-speed USB device number 3 using xhci_hcd
[  410.363415] usb 1-4: device descriptor read/64, error -71
[  410.599495] usb 1-4: device descriptor read/64, error -71
[  410.707509] usb usb1-port4: attempt power cycle
[  411.363359] usb 1-4: new full-speed USB device number 4 using xhci_hcd
[  411.363514] usb 1-4: Device not responding to setup address.
[  411.571534] usb 1-4: Device not responding to setup address.
[  411.779727] usb 1-4: device not accepting address 4, error -71
[  411.911830] usb 1-4: new full-speed USB device number 5 using xhci_hcd
[  411.911947] usb 1-4: Device not responding to setup address.
[  412.120132] usb 1-4: Device not responding to setup address.
[  412.327371] usb 1-4: device not accepting address 5, error -71
[  412.327456] usb usb1-port4: unable to enumerate USB device
```
```
sudo stm32flash -w nando_fw_v3_3_0.bin -v -g 0x0 /dev/ttyUSB0 

stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0414 (STM32F10xxx High-density)
- RAM        : 64KiB  (512b reserved by bootloader)
- Flash      : 512KiB (size first sector: 2x2048)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote and verified address 0x08040000 (100.00%) Done.

Starting execution at address 0x08000000... done.
```
Thanks for help.

bbogush, 2020-12-17:
Hi,
The NANDO fw should print logs via USART1, so you can check if fw was written successfully. 
Unfortunately cannot help with STM tool.

bra1nslayer, 2020-12-17:
Hi,
Just managed to work.I think microcontroller is executing code slow and failing to respond host pc.In meantime host was already giving up initing usb.
After flashing firmware i executed:
```
sudo stm32flash -g 0x08000000 /dev/ttyUSB0 

stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0414 (STM32F10xxx High-density)
- RAM        : 64KiB  (512b reserved by bootloader)
- Flash      : 512KiB (size first sector: 2x2048)
- Option RAM : 16b
- System RAM : 2KiB

Starting execution at address 0x08000000... done.
```
Then dmesg shows usb is working properly.
```
[ 4287.735237] usb 1-4: USB disconnect, device number 103
[ 4292.787484] usb 1-4: new full-speed USB device number 104 using xhci_hcd
[ 4292.965180] usb 1-4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[ 4292.965185] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4292.965188] usb 1-4: Product: STM32 Virtual COM Port  
[ 4292.965190] usb 1-4: Manufacturer: STMicroelectronics
[ 4292.965192] usb 1-4: SerialNumber: 56F967703335
[ 4293.029379] cdc_acm 1-4:1.0: ttyACM0: USB ACM device
```
Entered nando gui:
[image]
But after powercycling i always have to execute "sudo stm32flash -g 0x08000000 /dev/ttyUSB0 "
Also tried to flash fimrware from gui app.Same behavior.
Tried powering up board externally first then connected 2 usb pins sometimes works.
BTW:What does 0xEF 0xEF means?
Thanks.

bra1nslayer, 2020-12-17:
Tried another chip id detection seems work:
[image]

bbogush, 2020-12-17:
Hi,
It looks that fw is loaded from RAM, not flash. Have you removed BOOT0 pins connection?
ID 0xEF ...means that NAND memory is not connected or some high stray capacitance (usually due to not cleaned flux).

bra1nslayer, 2020-12-17:
Yes i have removed BOOT0 after flashing firmware.Same behavior if i don't execute  stm32flash -g 0x08000000 it doesn't recognize usb.I didn't specify filename in ""sudo stm32flash -g 0x08000000 /dev/ttyUSB0 "" i think it runs on flash not RAM.As i said "Also tried to flash fimrware from gui app.Same behavior." i think it proves i doesn't run on RAM right?I have never used STM32 before so i may be mistaken.

bbogush, 2020-12-17:
Yes, the firmware is written to flash. STM has different boot modes - from flash/ram. I've just had assumption that for some reason it starts code execution from RAM and ```sudo stm32flash -g 0x08000000 /dev/ttyUSB0``` makes it execute code from flash. If you can see serial port logs from firmware before command than it is not the case. It can be USB issue.

bra1nslayer, 2020-12-17:
Ok i see now, let me explain step by step what i have done:

1-Attached usb serial to board
2-Executed putty in /dev/ttyUSB0
3-Powered on board (Plugged in USB cable)
4-No output from putty
5-Executed `sudo stm32flash -g 0x08000000 /dev/ttyUSB0` on linux terminal
6-This serial output from putty:
```

Start application: 0

NAND programmer ver: 3.3.0
JTAG init...done.
LED init...done.
```

Is this an indication of my usb connection is bad?

Thanks for helping.

bbogush, 2020-12-17:
3-Powered on board (Plugged in USB cable) - After this step there should be logs so probably STM boots from RAM/System memory.
LED init...done. - there should be other logs after this, so there is also usb problem but maybe it caused by this switching from RAM boot to flash boot.
You can check boot settings according to stm manual
[image]

bra1nslayer, 2020-12-17:
I think i must use st-link since i don't have one, will order it.Is type works fine?
[image]

bbogush, 2020-12-17:
V2 should be ok.

bra1nslayer, 2020-12-17:
Thanks, just bought it.Will post results.

bra1nslayer, 2020-12-21:
St-link didn't arrived yet but i managed to read a Hynix HY27US08281A, here is the parameters i have used:
[image]

In csv format:HY27US08281A, 512, 16384, 16777216, 16, 0, 0, 0, 10, 10, 35, 25, 20, 10, 10, 10, 50, 50, 30, 2, 1, 0, -, 80, 144, 255, 128, 16, 96, 208, 112, 0, 173, 115, -, -, -
In gui format:HY27US08281A 0x200 0x4000 0x1000000 0x10 0 0 0 10 10 35 25 20 10 10 10 50 50 30 2 1 0x0 - 0x50 0x90 0xff 0x80 0x10 0x60 0xd0 0x70 0 0xad 0x73 - - -
Could you please add to database?

bbogush, 2020-12-21:
Hi,
I've added HY27US08281A to the BD.
HY27US08281A has only two bytes of ID but programmer tries to read five for all chips, thereby ID is repeating. It is ok, only minor UI presentation issue.
The case looks great. Thanks!

bra1nslayer, 2020-12-22:
Hi,
Here is version 3 of case:
https://www.dropbox.com/sh/j5aew40age1gje8/AADYzDvs2PvGYgaorMvxoTjaa?dl=0
I had to print 3 times to align connectors.
[image]
[image]
[image]
[image]
[image]
[image]
Tomorrow i will get st-link.Will share results.

bra1nslayer, 2020-12-23:
Hi,
St-link v2 is arrived and i have tried like below:I removed jumper from BOOT0.

```
st-info --probe
Found 1 stlink programmers
 serial: 483d0b002c135737334d4e00
openocd: "\x48\x3d\x0b\x00\x2c\x13\x57\x37\x33\x4d\x4e\x00"
  flash: 262144 (pagesize: 2048)
   sram: 65536
 chipid: 0x0414
  descr: F1 High-density device
```

```
st-flash write nando_fw_v3_4_0.bin 0x8000000
st-flash 1.6.0
2020-12-23T11:57:15 INFO common.c: Loading device parameters....
2020-12-23T11:57:15 INFO common.c: Device connected is: F1 High-density device, id 0x10036414
2020-12-23T11:57:15 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2020-12-23T11:57:15 INFO common.c: Attempting to write 262144 (0x40000) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0803f800 erased
2020-12-23T11:57:18 INFO common.c: Finished erasing 128 pages of 2048 (0x800) bytes
2020-12-23T11:57:18 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-12-23T11:57:18 INFO flash_loader.c: Successfully loaded flash loader in sram
128/128 pages written
2020-12-23T11:57:28 INFO common.c: Starting verification of write complete
2020-12-23T11:57:31 INFO common.c: Flash written and verified! jolly good!

```

Also tried flashing firmware with cube mx.
But still no luck.Device is not recognized.
Any suggestions?

bbogush, 2020-12-23:
Hi,
In order to double check this is not HW configuration problem could you check voltage level on pin 94 (BOOT0) and pin 37 (BOOT1) 
I am not aware of SW configuration that may cause boot from RAM but I will check later.

bra1nslayer, 2020-12-23:
Thanks for answer.
Pin 94 3.282v
Pin 37 zero(0) volt
I can provide all pin voltages on cpu if necessary.

bbogush, 2020-12-23:
Pin 94 - BOOT0 logical 1
Pin 37 - BOOT1 logical 0
According to this table it boots from "System memory". You should have 0 V on Pin 94 in order to boot from flash memory.
[image]

bra1nslayer, 2020-12-23:
**Yes it worked thank you!**
I was misconnected R15 to 3.3v instead of GND.What a mistake.
[image]

Dmesg output:

```
[ 3059.843292] usb 1-2: new full-speed USB device number 65 using xhci_hcd
[ 3060.017131] usb 1-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[ 3060.017136] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3060.017139] usb 1-2: Product: STM32 Virtual COM Port  
[ 3060.017141] usb 1-2: Manufacturer: STMicroelectronics
[ 3060.017142] usb 1-2: SerialNumber: 56F967703335
[ 3060.071236] cdc_acm 1-2:1.0: ttyACM0: USB ACM device

```
Serial output:
```

Bootloader ver: 3.4.0
Start application: 0

NAND programmer ver: 3.4.0
JTAG init...done.
LED init...done.
USB init...done.
CDC init...done.
Programmer init...done.
```

Issue is definetly solved but will test furthermore on different flash chips.Could you please don't close issue?

bbogush, 2020-12-23:
It's great that problem is found. I will not close issue. Let me know if you have problems with new 3.4.0 release.

bra1nslayer, 2020-12-23:
Thank you.
3.4.0 debian package seems dependecy problems or problem with my linux distro.
I have purged 3.3.0 from synaptic.libboost-system 1.67.0 is already installed.3.3.0 didn't have this problem.
[image]
[image]

bbogush, 2020-12-23:
I see binary is linked to specific boost library version. Your system has newer libraries.  I will try to fix this. Thanks for notifying.

bra1nslayer, 2020-12-23:
I have tested my hynix hy27us08281a on windows.Programmer detects it correctly but i have a question:
First i read chip and saved hynix1stread
Disconnected programmer from gui
Connected programmer from gui
Read chip again and saved file hynix2ndread
Compared 2 files using hex program:
[image]
[image]
It seems bytes are changing.Is this normal behavior?

bbogush, 2020-12-23:
Changing bytes are not normal, you should check connection from STM to NAND.

bra1nslayer, 2020-12-23:
Ok i have checked NAND to STM connections with multimeter no issues found.But I am not sure %100 of NAND chip because i salvaged it from a digital camera, so i have known working chip for testing purposes which is AM29LV160DB but couldn't find parameters such as tCLS tALS tCLH tALH etc.Any suggestions?
[AM29LV160DB.pdf](https://github.com/bbogush/nand_programmer/files/5737200/AM29LV160DB.pdf)
I have no experience about NAND parameters where can i learn these timings mean and where to look at datasheet because i think there are some synonym names for these timings i am pretty confused about parameters.
Hynix datasheets are seems a bit easy to find parameters but for example bb(bad block?)mark off.(offset?) this parameter where i should look in datasheet?
Sorry if i bothered you with this silly questions but if available a tutorial i want to learn how to fill these parameters from datasheets.

bbogush, 2020-12-24:
From my experience TSOP-48 socket does not provide reliable connection, especially if flash was unsoldered. Thereby I prefer soldering adapters.
AM29LV160DB is different kind of flash which is not compatible by pinout and parameters so is not supported.

bbogush, 2020-12-24:
Regarding parameters, there is no manual yet, I have plan to provide it soon, but they usually the same for chip family. 
"bb mark off." this is related to " Any block where the 6th Byte/ 1st Word in the
spare area of the 1st or 2nd page (if the 1st page is Bad) does not contain FFh is a Bad Block."
So this the offset of byte where bad block mark is set, the offset starts from 0, so in this case it will be 5 (6th byte).

bra1nslayer, 2020-12-24:
Thank you very much for explain.It will be very useful to have a guide, I will wait patiently.
Yes i checked AM29LV160DB pinouts are different.I found an old working router which have spansion s34ml01g100tf100 and i compared pinouts with HY27US08121B which seemed compatible:
[S34ML01G1-CypressSemiconductor.pdf](https://github.com/bbogush/nand_programmer/files/5739632/S34ML01G1-CypressSemiconductor.pdf)

[image]

After desoldering and cleaning flux and excess solder from spansion chip i installed it on socket.
I read spansion datasheet and i calculated sizes with my basic excel calculuator:esdf3
[NAND_CALCULUATOR.xlsx](https://github.com/bbogush/nand_programmer/files/5739639/NAND_CALCULUATOR.xlsx)
[image]
Then i filled up parameters as here:
S34ML01G1, 2048, 204800, 134217728, 128, 0, 20, 10, 10, 10, 10, 12, 12, 10, 5, 5, 5, 25, 25, 20, 3, 2, 0, 48, -, 144, 255, 128, 16, 96, 208, 112, 1, 241, 0, 29, 1
Checked read id, it reads id fine.
But when i try to read chip it gives error like below:
[image]
Seems like a parameter error right?

I also have noticed tsop socket is not realible, i agree soldering on pcb board will be bulletproof solution.
If i don't fit flash chip right it even cannot read id sometimes.

bbogush, 2020-12-24:
The parameters for S34ML01G1 should be as follows:
S43ML01G1, 2048, 131072, 134217728, 64, 0, 20, 10, 10, 10, 10, 12, 12, 10, 5, 5, 5, 25, 25, 20, 3, 2, 0, 48, -, 144, 255, 128, 16, 96, 208, 112, 1, 241, 0, 29, -
Your settings have only wrong block size and spare size.

bbogush, 2020-12-24:
I have similar flash - S43ML02G1. Just checked it should work for S43ML0xG1 chips.

bra1nslayer, 2020-12-24:
Yes!It worked and i compared 2 bin files with it and it matched.md5sum is also matching.So it was my hynix chip is bad.
Also binwalk recognizes image as jffs2 so i think it works perfectly.
Your parameters are correct.Thank you so much.
If it helps i can find another flash chips (i have a lot of junk electronic parts) and try to add more information for you with different chips.

bbogush, 2020-12-26:
Glad to hear it works. If you have spare flash chips it would very helpful to test them.

bra1nslayer, 2020-12-30:
I have desoldered many chips from old parts.Will test them tomorrow.

danyhm, 2021-01-04:
it's a driver issue for windows. I had this problem with another board. you need to replace the drivers using the [Zadig Tool](https://zadig.akeo.ie/) after that it's possible to upload the firmware using either St-link programmer or [dfu-util](http://dfu-util.sourceforge.net/)

> Hi,
> The best option is to use stlink programmer. You can try to use build-in USB/USART bootloader. For this you need to install stm32 tool https://www.st.com/en/development-tools/stm32cubeprog.html# [...]

danyhm, 2021-01-06:
> it's a driver issue for windows. I had this problem with another board. you need to replace the drivers using the [Zadig Tool](https://zadig.akeo.ie/) after that it's possible to upload the firmware [...]

According to AN2606 the DFU bootloader is not available in the f103 series. so I have to program with uart or st link as well.

bra1nslayer, 2021-01-07:
Hi again,
Sorry for late response, was very busy.I have scrapped many chips on old equipment.I have tested HY27US08**561**A which is similiar to HY27US08**281**A.Programer detected 561 chips as 281 because of first and second id.When i tried with 281 values it of course give an error because of size of chip.
[image]
So i decided to change 0x2000000 to 0x4000000 according to datasheet and it read perfectly.
[image]

[HY27US08561A_HynixSemiconductor.pdf](https://github.com/bbogush/nand_programmer/files/5783420/HY27US08561A_HynixSemiconductor.pdf)
[HY27US08281A_HynixSemiconductor(1).pdf](https://github.com/bbogush/nand_programmer/files/5783421/HY27US08281A_HynixSemiconductor.1.pdf)

Are there any methods to identify two different chips?
Will test other chips more.

danyhm, 2021-01-08:
I've assembled the PCB and programmed the STM32 and also could connect to the programmer

I expected it to give all 0xFF when no Nand is connected, however I'm getting 0x02 and sometimes 0x00 as the end byte

[image]

is this expected?

bbogush, 2021-01-08:
> Hi again,
> Sorry for late response, was very busy.I have scrapped many chips on old equipment.I have tested HY27US08**561**A which is similiar to HY27US08**281**A.Programer detected 561 chip as 281 because of first and second id.When i tried with 281 values it of course give an error because of size of chip.
> [image]
> 
> So i decided to change 0x2000000 to 0x4000000 according to datasheet and it read perfectly.
> 
> [image]
> 
> [HY27US08561A_HynixSemiconductor.pdf](https://github.com/bbogush/nand_programmer/files/5783420/HY27US08561A_HynixSemiconductor.pdf)
> [HY27US08281A_HynixSemiconductor(1).pdf](https://github.com/bbogush/nand_programmer/files/5783421/HY27US08281A_HynixSemiconductor.1.pdf)
> 
> Are there any methods to identify two different chips?

According to datasheet HY27US08281A and HY27US08561A have different IDs so maybe ID was not correctly read.
HY27US08561A - ADh 75h
HY27US08281A - ADh 73h

bbogush, 2021-01-08:
> I've assembled the PCB and programmed the STM32 and also could connect to the programmer
> 
> I expected it to give all 0xFF when no Nand is connected, however I'm getting 0x02 and sometimes 0x00 as the end byte
> 
> [image]
> 
> is this expected?

Pins do not have pull-up/down so when unconnected their value undetermined.

bbogush, 2021-01-09:
> According to AN2606 the DFU bootloader is not available in the f103 series. so I have to program with uart or st link as well.

Thank you for pointing this out. I was looking for this document to check.

danyhm, 2021-01-09:
I'm having problems reading a micron NAND flash

[image]

the part number is MT29F1G08ABAD which is very similar to MT29F4G08ABAD in it's parameters.

I can't get the programmer to detect the chip correctly. it's showing the above results

how can I start debugging this?

danyhm, 2021-01-09:
in fact having the NAND flash in the adapter or not doesn't make a difference in the results.

bbogush, 2021-01-09:
> also the pcb design is not the best. R2 is chosen as 10k which is too much and my board keeps resetting. i changed it to 1.5k and now the reset problem is solved

The general problems are:
- TSOP-48 socket does not provide reliable connection for unsoldered NAND because pins are not even so advise is use "soldering adapters", try to adjust NAND in socket to see the difference in read ID.
- no connection between MCU and adapter/NAND, try to check connections by tester.
- stay capacitance, make sure no flux leftovers after soldering.

danyhm, 2021-01-09:
after connecting and disconnecting all the wires I could get the first byte correct

[image]
[image]
[image]
[image]

but still no success

danyhm, 2021-01-09:
ok i managed to get it working. it seems to be a software issue.

I added the new chip in the parallel database and configured it according to the datasheet which is exactly the same as MT29F4G08ABAD except the chip ID is `0x2c 0xf1 0x80 0x95 0x02` and the total size is `0x8000000`

normally clicking detect gives the wrong output like 
```
Info: Detecting chip ...
Info: ID 0x2c 0x01 0x02 0x03 0x04
Info: ID 0xff 0xff 0xff 0xff 0x00
```
but if I select the chip manually and then click read , the chip is read. afterwards clicking detect gives the correct result
also when clicking read sometimes it gives the error `Error: Chip size is not set` while it actually is set in the database. but changing the totalsize to something else then solves the error and the chip can be read.

i hope I was clear in explaining the bug

danyhm, 2021-01-11:
just a question out of the topic.

I'm trying to restore my scope which firmware I corrupted by accident. I'm programming the bootloader after erasing the first block but after soldering the Nand it still doesn't boot. what could I be doing wrong? could it have to do something with the OOB data part? should that be cleared too ?

bbogush, 2021-01-11:
> just a question out of the topic.
>
> I'm trying to restore my scope which firmware I corrupted by accident. I'm programming the bootloader after erasing the first block but after soldering the Nand it [...]

I don't have MT29 to check but other chips work correctly. Have you verified that write was successful by comparing write and read dump? I still recommend to make soldering adapter to avoid any hw issues. OOB data should be already cleared by block erase because there is no separate command for OOB erase. The best approach to recover bootloader is to have dump from working device together with OOB.

danyhm, 2021-01-11:
> I don't have MT29 to check but other chips work correctly. Have you verified that write was successful by comparing write and read dump? I still recommend to make soldering adapter to avoid any hw [...]

yes I compared the written data with another read and the write has been successful. I have a dump of a working device but it doesn't have OOB data. I'm scratching my head here as logically it should work, but it's not working. :(

danyhm, 2021-01-12:
I think i have a lead.

The Zynq 7000 series in my oscope uses internal ECC feature of the Nand flash.
Since I'm programming with ECC disabled (default on POR) the OOB data doesn't include the parity bits generated by internal ECC and that might be why the bootloader is failing.

Nando software doesn't support enabling internal ECC yet , so I have 2 options
1- Implement the ECC feature my self. but I'm not familiar with the code flow and it takes a lot of time.
2- manually calculate the ECC parity bits and add them in OOB data and program with those.

do you know how the parity Bits are calculated ? 
it would be very neat if you could add a custom command window where one can send and receive custom data to the chips.

bbogush, 2021-01-12:
> I think i have a lead.
>
> The Zynq 7000 series in my oscope uses internal ECC feature of the Nand flash.
> Since I'm programming with ECC disabled (default on POR) the OOB data doesn't include the [...]

Good catch! I cannot help with ECC calculation but it's required feature and will think how to add it future.

danyhm, 2021-01-12:
thank you very much,
I read Micron datasheet again and it says the parity bit areas cannot be programmed manually.
although I'm not sure if that is the case when ECC is disabled. do you have any experience with this?

I think I'll have to implement the ECC enable command after all ...

bbogush, 2021-01-12:
> thank you very much,
> I read Micron datasheet again and it says the parity bit areas cannot be programmed manually.
> although I'm not sure if that is the case when ECC is disabled. do you have any experience with this?
> 
> I think I'll have to implement the ECC enable command after all ...

You can generate some binary file and write it to the first block including spare area. The read will show if this memory can be modified.

danyhm, 2021-01-12:
you are right, however, for that, I still need to know the algorithm used for the parity bit generation.
I found this from micron that explains their algorithm, but it doesn't match the one I have from the corrupted dump
https://www.micron.com/-/media/client/global/documents/products/technical-note/nand-flash/tn2963_ecc_in_slc_nand.pdf

I think It's better just to implement the enable ECC feature.

I'll start reading your code and will get back to you if I need help.

danyhm, 2021-01-15:
> I have tested my hynix hy27us08281a on windows.Programmer detects it correctly but i have a question:
> First i read chip and saved hynix1stread
> Disconnected programmer from gui
> Connected programmer [...]

I had this problem exactly and I think it's a software issue. I didn't track it down yet.
if you look closer you'll see you are having unexpected bytes appearing in your reading which causes the reads to be different. I think it to be a problem in USB communication to the host PC. it's either that or the packet handling. because the unexpected bytes usually appear at the beginning of each page. disconnecting the USB and connecting it again a couple of times usually solves the problem. also, I've had experiences with STM hanging in the middle of simple commands like ReadID. perhaps a hardfault?

also, the IDs appearing like `Info: ID 0x2c 0x01 0x02 0x03 0x04` are actually wrong and it's because the communication with NAND should start with a Reset command. I've implemented it and will make PR soon.

bbogush, 2021-01-15:
> I had this problem exactly and I think it's a software issue. I didn't track it down yet.
> if you look closer you'll see you are having unexpected bytes appearing in your reading which causes the [...]

I've got Micron NAND from a router and reproduced issue with ID. I see it requires reset after power on. At least it fixes chip detection. I will add fix tomorrow and will check different read buffer.

bbogush, 2021-01-30:
I see freshly unsoldered chip can have some bits changing during several reads but after erase-write the issue disappears so it may be related to chip itself.

bra1nslayer, 2021-02-09:
Hi all again,
I have ordered pcb from china and replicate project again.My dip version of nando was a bit tricky so i advise not to replicate dip verison of nando:)Anyway here is the photos if anyone interested:
[image]
[image]
[image]
Feel free to use photos in wiki or github if desired.
After making pcb i was able to read hy27us08281a chip without any issues at all.I think shorter cables are important in this case.Again please don't made my mistake (dip version) and  made a proper pcb for nando.
Now i am back to read and test some random chips but i have an issue with one of model which is K9K1G08U0M.It gives error:
[image]

`Error: "Programmer sent error: (113) Bad block table overflow. Probably some pins of chip are not connected to the programmer"`

I was using soldered method as picture above and cleaned pcb with ipa.
Chip id reads fine.I have 2 different chips same model (K9K1G08U0M) both chips give same error.
Here is my parameters:
[image]
[image]
I have filled these values according to wiki and K9K1G08U0M datasheet.
Any advise?
Regards.

bbogush, 2021-02-09:
Hi,
The board looks great. I think the problem might be that "Read 2 com." should be "-". This is a bit confusing because there is "Read 2" command but parameter actually means "2nd. Cycle" of "Read 1" command.
Also some other parameter is different for me, tAR=50, tREA=35.

[5 more replies]

closed 2021-02-28 by bbogush