bbogush/nand_programmer #26: Does it support STM32F407?

quanghau113, 2020-12-26:
I have a STM32F407 board. Can I make it work on STM32F4 series?

bbogush, 2020-12-26:
STM32F407 supports all required hardware features (FSMC, SPI, USB) but software supports only STM32F103 family (256-512MB flash).

enilfodne, 2021-01-22:
On that note, would you be willing to merge a PR for STM32F2/4 support?

It's a big PR (very hard to split up in smaller commits) and the scope is software-only (i.e. no new schematics or part-numbers).

bbogush, 2021-01-27:
> On that note, would you be willing to merge a PR for STM32F2/4 support?
> 
> It's a big PR (very hard to split up in smaller commits) and the scope is software-only (i.e. no new schematics or part-numbers).

I suggest to merge support of STM32F4 to separate branch such as support_stm32f4 and it may be integrated to master in future.

bbogush, 2021-02-28:
Closing old issues. Please reopen if open questions remain.

closed 2021-02-28 by bbogush