bbogush/nand_programmer #27: is it possible to erase by block instead of full chip erase?

danyhm, 2020-12-30:
Hello,

I'm in desperate need of a Nand programmer. I found yours which is opensource, before ordering the parts I want to know if it's possible to erase the blocks by their number instead of fully erasing the chip?

bbogush, 2020-12-30:
Hello,

Currently separate block erase is not supported.

danyhm, 2021-01-04:
ok, what if I change the chip DB and decrease the size of the chip?
I only need to erase the 1st block. it should work, right?

bbogush, 2021-01-04:
Yes, it will only erase specified size.

danyhm, 2021-01-04:
ok, thanks,
but for future versions, it would be nice to add this feature. other programmers like TNM or RT809 have an option where you can specify the start and end block.

nQ1PX, 2021-02-23:
I think this feature will be quite useful, given the read/write speeds are quite slow. It would be nice if you changed 10% of the flash content and then the programmer only erases and writes what's necessary. This is what flashrom does, it reads the chip, checks the provided binary and only erases and writes the differences.

With that said, we do have some problems that flashrom does not deal with. flashrom does read + write + verify if you use the write operation, but this is not really feasible in this project, due to how long it takes to read/write the entire flash. If the user swaps out flash after they read the chip, it's going to have a mismatch.

My proposal is to have a "delta write" option in the menu, so the user can use that if they know what they're doing and potentially save a lot of time. This would be *really* useful if someone just wants to change some environment variables or replace a bootloader on the flash and leave everything else intact.

Partial erases as a standalone feature is not all that important on its own imo, because this programmer is not really designed for permanent/regular use with a single chip, so one more erase cycle really isn't going to be too problematic on the endurance (plus most of the NAND we're targeting is SLC anyway).