bbogush/nand_programmer #30: Some UI/UX suggestions

nQ1PX, 2021-02-22:
Thanks for making this excellent project. There are some small UI/UX improvements that I think would be cool to have:
- a progress bar at the bottom of the UI for the current operation, along with a rough ETA display
- a menu option to halt current operation, this might need firmware support as well

bbogush, 2021-02-23:
Thanks for suggestions. Added to TODO list.

7134956, 2022-06-16:
Notification msgbox alert checkbox after completion.
https://github.com/7134956/nand_programmer/commit/7342c8347c822e09a18e2100c04b91efe89dc7f9
[image]   [image]

7134956, 2022-06-16:
Port selection from list. List updated on click.
https://github.com/7134956/nand_programmer/commit/cc86a70c1455b018681036f3e8ab0ff5f7985106
[image]
[image]

bbogush, 2022-06-18:
I've cherry picked commits. Great features. Thanks!

7134956, 2023-02-19:
https://github.com/7134956/nand_programmer/commit/d190b925c372fc0d8cae5c5a9d118296e4aa84fa
Add work area setup to GUI after chip select.
[image]

bbogush, 2023-02-26:
Cool feature. Merged. Thank you!

7134956, 2023-04-16:
https://github.com/7134956/nand_programmer/commit/78156aef862cf0129f2b0577b50faab9c5af4fa0
This can be merged with little changes for your branch.
[image]