bbogush/nand_programmer #31: NAND128-A/NAND256-A/NAND512-A/NAND01G-A: unsupported read command style

nQ1PX, 2021-02-23:
There is a series of very old NAND from STMicroelectronics:
[image]
Datasheet: https://datasheetspdf.com/datasheet/NAND128-A.html

The problem we have here is that they use 2 different commands for reading 2 different halves of a page, so with the current software setup, we can only really read the 1st half of the page. Adding support for these chips will require additional work on the firmware and the Qt host application.
[image]

Do you think it's worth the time making changes to the fw and host application to add support for these chips, or should we file these off as "impossible to support" for now?

bbogush, 2021-02-23:
Hello,
From what I understand the pointer can set to area A and read operation will read whole page including spare. So it should as others chips.
[image]

nQ1PX, 2021-02-23:
Oh, guess I didn't understand the datasheet very well then. I will try it with my chip and report back.

closed 2022-07-28 by bbogush