bbogush/nand_programmer #37: Parallel NOR flash support?

donnm, 2021-06-07:
I started adding support for M29W640GB in the chip database before realising it was NOR flash (couldn't find all the timing info in the datasheet). Does this have any chance of working?

bbogush, 2021-06-08:
Currently NOR is not supported, I think it will not work because NOR uses different programming approach.

gitercn, 2021-06-20:
I'm also looking for a programmer for Spansion S29 NOR FLASH programmer (TSOP56). Is there any open source NOR programmers? Could this project add NOR support?

bbogush, 2021-06-21:
> I'm also looking for a programmer for Spansion S29 NOR FLASH programmer (TSOP56). Is there any open source NOR programmers? Could this project add NOR support?

I haven't seen open source NOR programmers. This project can be extended to support NOR and STM chip has support of it. I'm not sure when it will be added because development requires some time.

donnm, 2021-06-21:
There's this, an Arduino Mega connected to a 32 pin PDIP NOR flash: https://mint64.home.blog/2018/07/29/parallel-nor-flash-eeprom-programmer-using-an-arduino-part-1-the-sst39sf040-and-planning/

Part 2 has the Arduino sketch code: https://mint64.home.blog/2018/07/30/parallel-nor-flash-eeprom-programmer-using-an-arduino-part-2-arduino-code-and-serial-comms/

closed 2023-05-23 by bbogush