bbogush/nand_programmer #41: NANDO Eagle PCB (48pin ZIF)

kooroshi60, 2021-08-22:
Here is an another variant of PCB for this great project. (Eagle 9 Format + PDF)
-Since the track clearance is .15mm, this PCB has to be made with Photo Resist film technic.
Toner transfer method is not precise enough.
-This is a double layer PCB so don't forget to solder VIAs.
-All the components except the USB connector are SMD.
-You can find the value of the components in the Eagle files.
-You can either use a 48pin ZIF IC Socket or 4 rows of pin headers in order to support 7.2mm and 15.24mm ZIFs.
-There are 18 pins of the socket which need a VIA wire. Other pins of ZIF socket and the USB connector are routed from Bottom layer.
[NANDO Bottom.pdf](https://github.com/bbogush/nand_programmer/files/7027975/NANDO.Bottom.pdf)
[NANDO TOP.pdf](https://github.com/bbogush/nand_programmer/files/7027976/NANDO.TOP.pdf)
[Nando_ZIF48_USBb.zip](https://github.com/bbogush/nand_programmer/files/7027980/Nando_ZIF48_USBb.zip)

bbogush, 2021-08-28:
Thanks for sharing ZIF schematic.
VET6 and VDT6 should also work, the redundant memory will not be used.

kooroshi60, 2021-12-24:
And here is a picture of completed Nando programmer. Many thanks to Mr. Bogus
[image]
h

bbogush, 2021-12-25:
Very nice PCB. Thank you!

ucsoom, 2023-03-19:
I created a programmer like yours based on your PCB file. The trouble now is that I have searched the internet for a suitable hex file for this programmer. Could you please send the hex file of your programmer? My email address is 401707682@qq.com

ucsoom, 2023-03-19:
[image]

bbogush, 2023-03-19:
@ucsoom all hex files are located on Google driver https://drive.google.com/drive/folders/1AGWV5_WANd6tItfXJFq4AlQePkRsusNU

ucsoom, 2023-03-20:
Thank you! Open Google's website here in China.

ucsoom, 2023-03-20:
Thank you! China has banned Google.

ucsoom, 2023-03-20:
The website you sent cannot be accessed

whitcloudfly, 2024-04-29:
可以用STM32cubeide开发hal库的固件，移植难度很小。

ucsoom, 2025-04-03:
My programmer doesn't know why, the Tsop48 chip is connected to the programmer, the programmer can't recognize it, and the Tsop48 chip programmer can recognize it. Can you help me?