bbogush/nand_programmer #46: Support for larger NAND chips?

7134956, 2022-06-19:
I see several restrictions that I have to work around.
uint32 is not enough for chip size.
Buffer needs to be moved to file.

7134956, 2022-06-19:
For 1GB chip, reading BB ended with timeout. I added  progress notification from  programmer for resolve issue. Progress displayed in user interface.
https://github.com/7134956/nand_programmer/commit/5774ef666753c6a2e69fba487edcee721c03e332

bbogush, 2022-06-19:
Thanks for BB read progress feature. I've merged it. 
I think chip size just need to be moved to uint64 but need to check. Could you elaborate your point about moving buffer to file?

7134956, 2022-06-19:
I don't know the architecture of the application, but the memory buffer for 1GB is not allocated . In my opinion, before reading the chip, you need to select a file and read into it. Viewer window must will display contents of the file.

7134956, 2022-07-01:
I modified the program to work without buffer allocation. Reading goes to the selected file with a request when overwriting. The file name is conveniently changed in the line or through the selection dialog.
[image]
https://github.com/7134956/nand_programmer/commit/1781d0974cb2c09b0408e690caab0ae6a80c527f

I also have code to support large chips, but I don't like that all the data in the table is now using uint64 and a lot of changes.

bbogush, 2022-07-10:
Great work! I've merged it. Thank you!

7134956, 2022-07-11:
QAbstractTableModel uses a lot of memory to support item selection. There is no way to change this behavior and refuse to allocate memory per row. I made a simple QAbstractScrollArea viewer widget and drew the content with a QPainter. The amount of memory does not grow with the size of the buffer. Commit will be posted soon.

bbogush, 2022-07-17:
Very nice solution with data viewer. Thank you!