bbogush/nand_programmer #59: Is K9F1G08U0D supported?

hpznot, 2023-04-16:
I added the K9F1G08U0D chip, but there was an warning:"Warning: QXcbConnection: XCB error: 3 (BadWindow), sequence: 14950, resource id: 13199545, major code: 40 (TranslateCoords), minor code: 0". And The following prompt appears when I connect.
"Info: Detecting chip ...
Info: ID 0xcf 0xcf 0xcf 0xcf 0xcf
Info: ID 0x80 0x80 0x80 0x80 0x02
Info: Chip not found in database
Error: Chip size is not set
Error: Chip size is not set
Info: Chip page size is unknown
Info: Reading bad blocks ...
Info: Bad blocks have been successfully read"
The above operations are detected,erased,read,written and read bad blocks in sequence.
[K9F1G08U0D-Samsung.pdf](https://github.com/bbogush/nand_programmer/files/11241958/K9F1G08U0D-Samsung.pdf)

deividAlfa, 2023-04-16:
Check
https://github.com/bbogush/nand_programmer/blob/master/qt/nando_parallel_chip_db.csv

Add a new entry for your chip.

hpznot, 2023-04-16:
> It seems supported: https://github.com/bbogush/nand_programmer/blob/master/qt/nando_parallel_chip_db.csv
> 
> However your ids seem strange. Maybe a bad connection?

The chip is soldered. I have also tried changing the chip, but it still doesn't work.

hpznot, 2023-04-19:
I found that I still have the same problem using K91G08U0E. Why is this?

hpznot, 2023-04-19:
Can 1117 output 3.28V be used?

> It seems supported: https://github.com/bbogush/nand_programmer/blob/master/qt/nando_parallel_chip_db.csv
> 
> However your ids seem strange. Maybe a bad connection?

Can 1117 output 3.28V be used?

deividAlfa, 2023-04-19:
The list has the K9F2G08U0A& M parts, but not the D, each one has a different ID.
The "D" ID is ```EC F1 00 95 40```

# name	page size	block size	total size	spare size	bad block mark off.	tCS	tCLS	tALS	tCLR	tAR	tWP	tRP	tDS	tCH	tCLH	tALH	tWC	tRC	tREA	row cycles	col. cycles	read 1 cycle com.	read 2 cycle com.	read spare com.	read ID com.	reset com.	write 1 cycle com.	write 2 cycle com.	erase 1 cycle com.	erase 2 cycle com.	status com.	set feat. com.	en. ECC addr	en. ECC val.	dis. ECC val.	ID1	ID2	ID3	ID4	ID5

K9G8G08U0D	2048	262144	1073741824	64	0	20	12	12	10	10	12	12	12	5	5	5	25	25	20	3	2	0	48	-	144	255	128	16	96	208	112	-	-	-	-	236	241	00	149	64

closed 2023-05-23 by bbogush