bbogush/nand_programmer #70: add NAND Micron 29F1G08ABADA

web-rpi, 2023-10-05:
Please let me know if I have correctly added the chip Micron 29F1G08ABADA.
[micron_technology_micts05940-1-1759186.pdf](https://github.com/bbogush/nand_programmer/files/12825285/micron_technology_micts05940-1-1759186.pdf)

Here is my current parallel chip database:
[nando_parallel_chip_db.csv](https://github.com/bbogush/nand_programmer/files/12825293/nando_parallel_chip_db.csv)

The chip is recognized, but when I attempt to read, it fails. If I set 'First Block' to 1 and 'Last Block' to 2047, the chip is read. However, the default values are 0 and 8191 once the chip is detected.

[image]

Thank you!

bbogush, 2023-10-09:
Looks "total size" is not correct, should be 1 Gb  * 1024 Mb * 1024 Kb  * 1024  / 8 bit = 134217728

web-rpi, 2023-10-12:
Oh yes! You're right! I corrected this value and it works :)
1GB != 1Gb .... 

Thank you!

closed 2023-10-15 by bbogush